DFRobot/DFRobotDFPlayerMini #7: readState()

9H1LO, 2017-10-25:
Serial.println(myDFPlayer.readState());

sometimes while playing returns 513 or stopped 512

but most of the time returns -1 or some other number unless rebooted

using esp8266

haaslukas, 2017-11-09:
I'm having a similar issue: When I have this line in the code...
`  if (!myDFPlayer.readState()){
     // do something if the mp3 has ended
  }`
my esp8266 stops working after approx. 10 to 20 minutes. I also saw these 513 or 512 returns when debugging, but could never figure out what is actually stopping or sometimes resetting the esp8266.

any ideas?

sadfsdfsdaf, 2018-05-29:
Guys, take a look at this issue https://github.com/DFRobot/DFRobotDFPlayerMini/issues/12
I think that the problem can be common for both issues. This library uses uninitialized variables inside the response parsing handler. It means that sometimes valid responses can be interpreted using statuses from the previous commands.

jimaobian, 2018-08-15:
Hi, the latest update may fix this problem. Try it here:
https://github.com/DFRobot/DFRobotDFPlayerMini/archive/1.0.3.zip

9H1LO, 2019-03-01:
there is still something wrong, i have also tried with 1.0.3 and 1.0.5

you set the player to loop play a track and in the main loop, every 5 seconds call getVolume 

in main loop:
`  static unsigned long timer = millis();
  if (millis() - timer > 5000) {
    timer = millis();
   getVolume();
}

void getVolume(){
    volume = mp3.readVolume();
    Serial.print("    Volume: ");
    Serial.println(volume);
}
`

the result is:

10:33:50.323 -> PLAY: Track:40
10:33:53.397 ->     Volume: 1
10:33:58.396 ->     Volume: 40
10:34:03.389 ->     Volume: 1
10:34:08.406 ->     Volume: 40

its as is the library is using some shared variables or interpreting the player's responses wrongly

with readFileCounts and readState the results are even worst also

9H1LO, 2019-03-01:
after doing some research i have read that these are symptoms of "fake" (aliexpress type) dfplayer modules, i didn't realize these could even exist for such a simple mp3 player (why bother??) maybe fake or chips which failed QC ? anyway, if someone has experience with this pls advise and close this thread as the issue is probably due to fake module, not the library. Hopefully this info will also help others with issues! I have now ordered a module directly from DFRobot website to confirm all this

imranaalam, 2019-10-25:
@9H1LO was the assumption of fake dfplayers not behaving as expected is valid?

9H1LO, 2019-12-30:
@imranaalam i received the module from DFRobot but have not got round to testing it and comparing

nicholaskillin, 2022-05-20:
Has anyone actually sorted this out? I have a DFPlayer that doesn't have this problem, but I've shared my project with others and some of them are running into this. I've confirmed that they are getting `512` as the response to `.readState()` and I'm really confused why some people run into this and others don't.

jasonglisson, 2022-06-07:
> Has anyone actually sorted this out? I have a DFPlayer that doesn't have this problem, but I've shared my project with others and some of them are running into this. I've confirmed that they are [...]

Just curious what project it is that you're having people comment about this issue. I am getting these 512 errors on my current project and was wondering if I could see what your project users are experiencing to see if I can troubleshoot. I suspect that the dfplayer is reseting because of a voltage spike or something.

nicholaskillin, 2022-06-07:
> Just curious what project it is that you're having people comment about this issue. I am getting these 512 errors on my current project and was wondering if I could see what your project users are [...]

My project can be found here:

https://github.com/nicholaskillin/baby-r2-sounds

That should have all of the documentation and code, including wiring diagrams. If you happen to find something I'd love to hear about it!

For now it seems like it's dependent on the DFPlayer itself. I've purchased 4 now from HiLetgo off Amazon and I've not had any trouble using this library with that specific model. However I had someone else reach out for help troubleshooting. Found out they were getting `512` from their players. They purchased theirs directly from the DFRobot and one from Digikey.

Players that worked: 
- https://www.amazon.com/gp/product/B01D1D0E7Q/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1

Players that don't work:
- https://www.dfrobot.com/product-1121.html?fbclid=IwAR33kexA3I6MpVHscI2PRaD_InpwsOrJ6F9iEBH7Y8vIuZ0lATHGkUS48Yo
- https://www.digikey.com/catalog/en/partgroup/dfplayer-a-mini-mp3-player-for-arduino/70505?fbclid=IwAR2to-wINysAqCk0twiXt-R2sz3DPyJSHhC2aF0ho9VHiIZlleqYmOPV7IY

jasonglisson, 2022-06-07:
> My project can be found here:
>
> https://github.com/nicholaskillin/baby-r2-sounds
>
> That should have all of the documentation and code, including wiring diagrams. If you happen to find something I'd [...]

Interesting. I've bought about 15 from Digikey exclusively and haven't had an actual issue with any of them. But......what I'm seeing as an issue, I think is actually a power related problem. 

You will see all over the internet that Dfplayer runs up to 5v, but that's actually too high. You might be running a project at 5v, but in reality you are likely getting 4.2v or something like that. 

With my project I'm using a booster board to get up to 5v (or very very close to it) and up to 2amps and the Dfplayer keeps resetting. So I put a diode between the power source and the Dfplayer to knock it down to 4v and it is more stable now.....but not completely. I think adding a capacitor will help. 

When the Dfplayer turns on at full volume, that's usually when it resets.

jasonglisson, 2022-06-07:
> > Interesting. I've bought about 15 from Digikey exclusively and haven't had an actual issue with any of them. But......what I'm seeing as an issue, I think is actually a power related problem.
> > You will see all over the internet that Dfplayer runs up to 5v, but that's actually too high. You might be running a project at 5v, but in reality you are likely getting 4.2v or something like that.
> > With my project I'm using a booster board to get up to 5v (or very very close to it) and up to 2amps and the Dfplayer keeps resetting. So I put a diode between the power source and the Dfplayer to knock it down to 4v and it is more stable now.....but not completely. I think adding a capacitor will help.
> > When the Dfplayer turns on at full volume, that's usually when it resets.
> 
> Hm, that is interesting. According to the tech specs it can operate on as little as 3.3V. I haven't measured, but in theory I'm sending mine about 4.8V. It would be interesting for sure to see if dropping it to 4V and adding a capacitor would help. A quick search on some other forums seems to suggest that you very well may be correct. Most people recommending to run it under 5V.
> 
> I don't have one of the non-working models to test with (got lucky my first time purchasing), so if you happen to get it stable I'd love to hear how you hooked it up.

Believe me, you'll hear from me when I get this figured out. I've been working on it for about 8 months now and this is my only hangup now. Dfplayer is just a nightmare. My project has about 5 different PCBs that are all connected. The main PCB has a Nano, Dfplayer, two shift registers, a servo, and a bluetooth connector. It's a prop for Ghostbusters. :-)

nicholaskillin, 2022-06-08:
> Believe me, you'll hear from me when I get this figured out. I've been working on it for about 8 months now and this is my only hangup now. Dfplayer is just a nightmare. My project has about 5 [...]

First of all, I'd love to see your ghostbusters prop when it's done. 

Second of all, this conversation got me thinking about the problem again. As far as I can tell, all readState() does is give you the status of the mp3. It may provide detail that you need, but all I needed to know was when the last mp3 file played was done playing. 

In looking at the documentation I realized that I completely overlooked the 'busy' pin on the player. 

I just updated my code to look at and read that state instead and it works great. 

Sent the updated code to the person that was having trouble with the first party DFPlayers and those players now work for him as well. 🎉

Again, you may need more detail and there may still be power issues worth solving, but wanted to pass that info along in case it's helpful for you.