DFRobot/DFRobotDFPlayerMini #9: Failure to initialise

renegadeandy, 2018-02-06:
Please check my question here : https://stackoverflow.com/questions/48650505/audio-playback-module-on-esp8266-fails-to-initialise

jimaobian, 2018-08-15:
Hi try the latest version 1.0.3.
https://github.com/DFRobot/DFRobotDFPlayerMini/archive/1.0.3.zip

rolfbachmann, 2018-08-30:
I tried the GetStarted example today using the latest version (1.0.5) through PlatformIO. I had to change the `myDFPlayer.begin(mySoftwareSerial)` call to `myDFPlayer.begin(mySoftwareSerial, false)`, so the isAck is set to false. This is described in link from the first comment too.

After changing that, everything works fine.

MauiJerry, 2018-10-23:
I have tried with Esp8266, Esp32, uno, mega and teensy. All of them fail if we use the default ACK Required.  However, if we specify False (as Rolf mentions) we get past this initialization and every thing works ok.  
Note however, that if we do so, begin() will succeed even if no player is connected.
So that status is utterly useless.  Might as well remove the code from examples and make it a void function.

iZhangHui, 2018-11-12:
@MauiJerry @rolfbachmann  Could you help to verify my PR #19 ?

rolfbachmann, 2018-11-12:
Tried out. Use `myDFPlayer.begin(mySoftwareSerial)` as it's used in the GetStarted.ino
V1.0.5: Failed
With your patch applied: Work

But it works only after warm boot (after upload/reset button), not after cold boot (power off and on). Don't know if it's my wire setup.

Edzelf, 2019-07-24:
The time-out for the reply on the reset command is too short (500 msec).  You may use:
`myDFPlayer.setTimeOut ( 1000 ) ;`
just before the call to myDFPlayer.begin().
With my player the reply comes after about 850 msec.

ESPboy-edu, 2019-10-20:
ESP8266, lib v1.0.5 still ACK problem. even with myDFPlayer.setTimeOut ( 1000 ) ;
just before the call to myDFPlayer.begin() no ACK
have to use myDFPlayer.begin(Serial, false) during init to proceed and it works fine after, but no way to understand is mp3 module connected or not (

Edzelf, 2019-10-20:
You can't use "Serial" for the communication to the DFPLayer.
Use myDFPlayer.begin(mySoftwareSerial) ;
See for example [here](http://arduiner.blogspot.com/2016/06/nodemcu-esp8266-dfplayer-mp3-music-from.html) and add the "myDFPlayer.setTimeOut ( 1000 ) ;" just before the call to myDFPlayer.begin().
D1 and D2 are used for RX and TX in this example.

ESPboy-edu, 2019-10-22:
why i can't use "Serial"? )
i have no free pins for software serial ))
http://community.espboy.com/t/espboy-description-and-discussion/20

anyway with software serial is the same problem
i can drive DFPLayer fine but can't get info from it
volume, eq, mode, fileno... nothing.
always "timeout" even with 10000ms ))

but with the lib https://github.com/Makuna/DFMiniMp3 i can read )
but it does not work with "Serial" and initialization  has no info is it done or no

Edzelf, 2022-03-09:
The serial output of an Arduino Uno is too high for the dfplayer.  Use a 1k resistor in series.

alfem, 2022-03-12:
Did anyone tested this specific fork of softwareserial?

https://github.com/plerup/espsoftwareserial

o15a3d4l11s2, 2024-05-18:
After wiring the DFPlayer module to ESP32, I decided to not connect the speaker, but just to rely on the LED indicator of the module to understand if an audio file was playing.

This was causing the described issue for me - not being able to initialise the module, unless I set `isACK` to `false`.
Once I wired the speaker, everything was fine and there was no need to set `isACK = false`.

Just sharing, maybe someone else also faces the same issue.