DFRobot/DFRobotDFPlayerMini #10: HardwareSerial on Mega 2560

houbig, 2018-02-27:
Hi, please help me to understand what's going on. When I'm using the SoftwareSerial communication between DFPlayer Mini and Arduino Mega 2560 everething works perfect but when I'm trying to use the HardwareSerials Serial1, Serial2 or Serial3 I'm in the stupid condition

_Unable to begin:
1.Please recheck the connection!
2.Please insert the SD card!_

My initialization is following

`#include "Arduino.h"
#include "DFRobotDFPlayerMini.h"

DFRobotDFPlayerMini myDFPlayer;
void printDetail(uint8_t type, int value);

void setup()
{

  Serial.begin(9600);
  Serial1.begin(9600);
  
  
  Serial.println();
  Serial.println(F("DFRobot DFPlayer Mini Demo"));
  Serial.println(F("Initializing DFPlayer ... (May take 3~5 seconds)"));
  
  if (!myDFPlayer.begin(Serial1)) {  //Use softwareSerial to communicate with mp3.
    Serial.println(F("Unable to begin:"));
    Serial.println(F("1.Please recheck the connection!"));
    Serial.println(F("2.Please insert the SD card!"));
    while(true);
  }
  Serial.println(F("DFPlayer Mini online."));
  
  myDFPlayer.setTimeOut(500); //Set serial communictaion time out 500ms
  
  //----Set volume----
  myDFPlayer.volume(10);  //Set volume value (0~30).
  
  //----Set different EQ----
  myDFPlayer.EQ(DFPLAYER_EQ_NORMAL);
  
  //----Mp3 play----
  myDFPlayer.loop(1);  //Loop the first mp3
  
  //----Read imformation----
  Serial.println(myDFPlayer.readState()); //read mp3 state
  Serial.println(myDFPlayer.readVolume()); //read current volume
  Serial.println(myDFPlayer.readEQ()); //read EQ setting
  Serial.println(myDFPlayer.readFileCounts()); //read all file counts in SD card
  Serial.println(myDFPlayer.readCurrentFileNumber()); //read current play file number
  Serial.println(myDFPlayer.readFileCountsInFolder(3)); //read fill counts in folder SD:/03
}`

Please help me to understand what I'm doing wrong or within your library I can't use the HardwareSerial?

houbig, 2018-03-01:
Ok, problem comes from the Arduino Mega board.

closed 2018-03-01 by houbig

pschmid1502, 2019-03-16:
Hi @houbig,
currently, I have the same problem you mentioned. How did you solved it?
Thank you for your answer!

houbig, 2019-03-17:
Hi, I’ve changed the arduino board.

ysmianji, 2019-11-08:
problem is TimeOut, so:

Serial.begin(9600);
Serial1.begin(9600);

Serial.println();
Serial.println(F("DFRobot DFPlayer Mini Demo"));
Serial.println(F("Initializing DFPlayer ... (May take 3~5 seconds)"));

**myDFPlayer.setTimeOut(1000); //Set serial communictaion time out 1000 ms**

if (!myDFPlayer.begin(Serial1)) { //Use softwareSerial to communicate with mp3.
Serial.println(F("Unable to begin:"));
Serial.println(F("1.Please recheck the connection!"));
Serial.println(F("2.Please insert the SD card!"));
while(true);
}