DFRobot/DFRobotDFPlayerMini #20: Noise is inserted into audio during communication (It's playing?)

rtek1000, 2018-11-15:
I would like to monitor the status every 1 second to see if it is playing.

I have a folder with files of small duration and others with long files, but some folders can not be played, so I can not use the continuous play commands, or play randomly, for the whole card.

I modified the example file, to randomly play some folders, but it works fine only if I use the busy pin.

I noticed that a noise appears in the audio, when the serial port is used, so I had to monitor the busy signal pin.

I've added series resistors for RX and TX, and the noise has decreased a lot, but sometimes the program reports errors such as "Time Out!", "Stack Wrong!",

How to monitor by software without causing a noise similar to the step of a wall clock in the audio?

`/***************************************************
  DFPlayer - A Mini MP3 Player For Arduino
  <https://www.dfrobot.com/product-1121.html>

 ***************************************************
  This example shows the all the function of library for DFPlayer.

  Created 2016-12-07
  By [Angelo qiao](Angelo.qiao@dfrobot.com)

  GNU Lesser General Public License.
  See <http://www.gnu.org/licenses/> for details.
  All above must be included in any redistribution
 ****************************************************/

/***********Notice and Trouble shooting***************
  1.Connection and Diagram can be found here
  <https://www.dfrobot.com/wiki/index.php/DFPlayer_Mini_SKU:DFR0299#Connection_Diagram>
  2.This code is tested on Arduino Uno, Leonardo, Mega boards.
 ****************************************************/

#include "Arduino.h"
#include "SoftwareSerial.h"
#include "DFRobotDFPlayerMini.h"

#define busyPin 12;

SoftwareSerial mySoftwareSerial(5, 4); // RX, TX
DFRobotDFPlayerMini myDFPlayer;
void printDetail(uint8_t type, int value);

bool isPlaying = false;

int lastValidFolder = 0;
const int maxFolder = 10; // (1~99)
int folderFiles[maxFolder];

void setup()
{
  pinMode(busyPin, INPUT); // optional
  
  randomSeed((analogRead(6) + analogRead(7)) / 2); //

  mySoftwareSerial.begin(9600);
  Serial.begin(9600);

  Serial.println();
  Serial.println(F("DFRobot DFPlayer Mini Demo"));
  Serial.println(F("Initializing DFPlayer ... (May take 3~5 seconds)"));

  //Use softwareSerial to communicate with mp3.
  if (!myDFPlayer.begin(mySoftwareSerial)) {
    Serial.println(F("Unable to begin:"));
    Serial.println(F("1.Please recheck the connection!"));
    Serial.println(F("2.Please insert the SD card!"));
    while (true);
  }
  Serial.println(F("DFPlayer Mini online."));
[... 301 lines cut ...]
          Serial.println(F("Check Sum Not Match"));
          break;
        case FileIndexOut:
          Serial.println(F("File Index Out of Bound"));
          break;
        case FileMismatch:
          Serial.println(F("Cannot Find File"));
          break;
        case Advertise:
          Serial.println(F("In Advertise"));
          break;
        default:
          Serial.print(F("Unknown, "));
          Serial.println(value, DEC);
          break;
      }
      break;
    default:
      Serial.print(F("Unknown type code, "));
      Serial.println(type, DEC);
      break;
  }
}`

[image]

rtek1000, 2018-11-15:
When the "Time Out!" error appears, the song is stopped before it ends, and another is started,
(This occurs when the busy pin is not used)

Log result using the busy pin:
> Host - Play file 6 from folder 3
> DFPlayer - Play Finished!
> Host - Play file 5 from folder 3
> DFPlayer - Play Finished!
> Host - Play file 63 from folder 1
> DFPlayer - Play Finished!
> Host - Play file 5 from folder 4
> DFPlayer - Play Finished!
> Host - Play file 4 from folder 3
> DFPlayer - Stack Wrong!
> DFPlayer - Play Finished!
> Host - Play file 18 from folder 2
> DFPlayer - Play Finished!
> Host - Play file 21 from folder 6
> DFPlayer - Play Finished!
> DFPlayer - Play Finished!
> Host - Play file 4 from folder 3
> DFPlayer - Play Finished!
> Host - Play file 17 from folder 5
> DFPlayer - Play Finished!
> Host - Play file 2 from folder 5
> DFPlayer - Stack Wrong!
> DFPlayer - Play Finished!
> Host - Play file 11 from folder 4
> DFPlayer - Play Finished!
> Host - Play file 2 from folder 4
> DFPlayer - Play Finished!
> Host - Play file 2 from folder 4
> DFPlayer - Play Finished!

Result without using busy pin:
> Host - Play file 99 from folder 1
> DFPlayer - Play Finished!
> Host - Play file 98 from folder 1
> DFPlayer - Play Finished!
> Host - Play file 78 from folder 1
> DFPlayer - Time Out!
> Host - Play file 61 from folder 1
> DFPlayer - Play Finished!
> Host - Play file 80 from folder 1
> DFPlayer - Play Finished!
> Host - Play file 46 from folder 1
> DFPlayer - Play Finished!
> Host - Play file 8 from folder 1
> DFPlayer - Time Out!
> Host - Play file 84 from folder 1
> DFPlayer - Play Finished!
> Host - Play file 104 from folder 1
> DFPlayer - Error: Cannot Find File
> Host - Play file 17 from folder 1
> DFPlayer - Time Out!
> Host - Play file 88 from folder 1
> DFPlayer - Play Finished!
> Host - Play file 38 from folder 1
> DFPlayer - Play Finished!
> Host - Play file 105 from folder 1
> DFPlayer - Time Out!
> DFPlayer - Error: Cannot Find File
> Host - Play file 5 from folder 1
> DFPlayer - Play Finished!
> DFPlayer - Play Finished!
> Host - Play file 56 from folder 1
> DFPlayer - Play Finished!
> Host - Play file 69 from folder 1
> DFPlayer - Play Finished!
> Host - Play file 11 from folder 1
> DFPlayer - Time Out!
> Host - Play file 103 from folder 1
> DFPlayer - Error: Cannot Find File
> Host - Play file 58 from folder 1
> DFPlayer - Play Finished!
> Host - Play file 91 from folder 1
> Host - File playback time 1s
> Host - Play file 76 from folder 1
> DFPlayer - Unknown type code, 11
> Host - Play file 34 from folder 1
> DFPlayer - Play Finished!
> Host - Play file 19 from folder 1
> DFPlayer - Play Finished!
> Host - Play file 93 from folder 1
> DFPlayer - Time Out!
> Host - Play file 5 from folder 1
> DFPlayer - Play Finished!
> DFPlayer - Play Finished!

drolsen, 2018-11-17:
I'd recommend adding a small 10uf capacitor on your vcc/grn lines to help pool up some dedicated mA to the DFRobot. Strange noises can happen when your microcontroller(s) draw mA away from the mini player without one.

rtek1000, 2018-11-21:
I managed to make the code more stable, I noticed that the x routine shows the value 512 for the stopped state. I no longer need to use the busy pin.

I updated the code, in case someone wants to try to run this.

This should generate a log on the serial (USB) port that looks like this:

> DFRobot DFPlayer Mini Demo
> Initializing DFPlayer ... (May take 3~5 seconds)
> DFPlayer Mini online.
> State: 512
> Volume (030): 30
> Equalizer (Normal/Pop/Rock/Jazz/Classic/Base): Normal(0)
> File Counts: 438
> Current File Number of micro SD Card: 0
> File Counts In Folder 1: 111 [Valid]
> File Counts In Folder 2: 54 [Valid]
> File Counts In Folder 3: 12 [Valid]
> File Counts In Folder 4: 20 [Valid]
> File Counts In Folder 5: 59 [Valid]
> File Counts In Folder 6: Not found
> File Counts In Folder 7: Not found
> File Counts In Folder 8: Not found
> File Counts In Folder 9: Not found
> File Counts In Folder 10: Not found
> Host - Play file 4 from folder 3
> ....5s....10s....15s....20s....25s....30s
> ....35s....40s....45s....50s....55s....1m:0s
> ....1m:5s....1m:10s....1m:15s....1m:20s....1m:25s....1m:30s
> ....1m:35s....1m:40s....1m:45s....1m:50s....1m:55s....2m:0s
> ....2m:5s....2m:10s....2m:15s....2m:20s....2m:25s....2m:30s
> ....2m:35s....2m:40s....2m:45s....2m:50s...
> DFPlayer - Play Finished!
> 
> DFPlayer - Play Finished!
> .Host - File playback time 2m:55s
> Host - Play file 23 from folder 1
> .
> DFPlayer - Play Finished!
> 
> DFPlayer - Play Finished!
> Host - File playback time 2s
> Host - Play file 23 from folder 5
> ....5s....10s....15s....20s....25s....30s
> ....35s....40s....45s....50s....55s....1m:0s
> ....1m:5s....1m:10s....1m:15s....1m:20s....1m:25s....1m:30s
> ....1m:35s....1m:40s....1m:45s....1m:50s....1m:55s....2m:0s
> .
> DFPlayer - Play Finished!
> 
> DFPlayer - Play Finished!
> Host - File playback time 2m:2s
> Host - Play file 63 from folder 1
> .
> DFPlayer - Play Finished!
> 
> DFPlayer - Play Finished!
> Host - File playback time 2s
> Host - Play file 5 from folder 3
> ....5s....10s....15s....20s....25s....30s
> ....35s....40s....45s....50s....55s....1m:0s
> ....1m:5s....1m:10s....1m:15s....1m:20s....1m:25s....1m:30s
> ....1m:35s....1m:40s....1m:45s....1m:50s....1m:55s....2m:0s
> ....2m:5s....2m:10s....2m:15s....2m:20s....2m:25s....2m:30s
> ....2m:35s....2m:40s....
> DFPlayer - Play Finished!
> 
> DFPlayer - Play Finished!
> 2m:45sHost - File playback time 2m:46s
> Host - Play file 2 from folder 3
> ....5s....10s....15s....20s....25s....30s
> ....35s....40s....45s....50s....55s....1m:0s
> ....1m:5s....1m:10s....1m:15s....1m:20s....1m:25s....1m:30s
> ....1m:35s....1m:40s....1m:45s....1m:50s....1m:55s....2m:0s
> ....2m:5s....2m:10s....

closed 2018-11-21 by rtek1000