DFRobot/DFRobotDFPlayerMini #22: Arduino Library Reference?

markpos, 2019-02-01:
Is there a reference guide to the Arduino library please? I can work out most of the functions from the example code, but a few things are eluding me! Been unable to find anything through search.

Many thanks!

terrypin999, 2022-01-25:
Me too!

Two years on ... did you ever find one?

rtek1000, 2022-06-06:
It might be interesting to read these documents:

https://picaxe.com/docs/spe033.pdf

https://doc.riot-os.org/group__drivers__dfplayer.html#details

terrypin999, 2022-06-06:
Thanks. In the following 4 months or so I did find the first, which I recall was helpfu. The second baffles me. It seems to about a set of ‘drivers’, not rh DFR library. And uses commands/functions with very different names.

rtek1000, 2022-06-07:
>  ... It seems to about a set of ‘drivers’, not the DFR library. And uses commands/functions with very different names.

Yes, but for that even though it can be an alternative source of information, it also has data there, and as the DFPlayer Mini datasheet is really very limited, I think it helps to have more points of view.

[image]
[image]
[image]

terrypin999, 2022-06-07:
I agree and I’m very keen to learn how I can use those potentially superior commands. I’m heavily invested in coding work-arounds for the (awful) DFR Mini library to complete my music box project. This must use the same hardware as my preceding project  for which I successfully used DFR’s other library for their USB module. (Motivation: micro-SD has a 32 GB capacity v 128 MB for the USB.)

I’m not a trained programmer, just a novice ‘Arduino programmer’. So Arduino only, no advanced C/C++ . Therefore, although I would have ideally liked to use what I call the ‘execute hex commands’ programs I’ve found, I still have not found an understandable (commented) example of exactly how to do that. At least, not for the sort of operations I need. Essentially, detecting a button low and jumping to a folder and track, printing that folder and track, looping or randomly playing a folder, and similar.

As soon as I get the DFR library program finished I aim to tackle it afresh. Do you have or know of a simple program that could help me overcome my mental block to get started on that please?

rtek1000, 2022-06-07:
The Arduino platform is limiting by itself, the concept is to always work at a high level and not have to worry about the hardware. Which in reality is not always true. There are many libraries for simple things, and few libraries for complex things. An example of this is the use of USB in Host mode. There are some very good libraries that were written for specific hardware, but are difficult to port to other hardware.

One thing that can help someone starting out is knowing the hardware, reading the datasheet and understanding what is written there, if you don't understand something, seek knowledge to understand.

Knowing the hardware can be a good starting point, starting with a simpler microcontroller (uC) like the PIC12F629 (which does not have an analog-to-digital converter) and doing assembler programming (ASM), this way you will understand that they are simple instructions that make things work.

The datasheet of a PIC12F629 is much simpler than the datasheet of an ATmega328, which was initially written by Atmel, later adapted by Microchip, but there is also the difference in resources between these two microcontrollers.

There is something called set of instructions (instruction set), which are the simplest commands that are used to make the hardware work, in the PIC12F629 there are 35 instructions. The ATmega328 has 131 instructions. On an ARM microcontroller it's much more, for an STM32F407 it's about 246 instructions.

The programmer who doesn't know the hardware is dependent on the libraries. In other words, the libraries may not offer a good use of the hardware, it will depend on how these libraries were written, many of them have redundant codes, which could be optimized.

A programmer who knows the hardware can easily understand how the library was written, because he knows what to expect from the library, the result that the library has to deliver is just the interface between the high-level language (C/C++/Basic/Pascal etc) and low-level language (assembler instructions, like those 35 instructions for the PIC12F629).

On the other hand, those who don't want to know the hardware are those people who live superficially, either because they don't have time, or are not willing to learn new things (if they have time). High-level languages help a lot to reduce project development time, and there is even an IDE (integrated development environment) that makes the code visually, using the mouse to create a block of code or a window that will be displayed. on a display, for example.

So, in short, the programmer has to dedicate himself to understanding how things work, but this takes time and the programmer needs to know how to manage time. There are courses to know how to manage time, it can be a good starting point too.