DFRobot/DFRobotDFPlayerMini #30: Example code doesnt seem to work on a Seeeduino XIAO?

ssurovich, 2020-12-10:
I needed a small Arduino for a project and I went with the Seeeduino XIAO which is a great little processor - but the same code that works on my other Arduinos does not execute correctly on the Seeeduino (or so it appears).  I am new to the Seeeduino offering, so it may be something obvious that I'm missing.

Its acting like it locks up - The code uploads and seems to start, but the triggers arent working to start the first MP3 on the SD - However the same circuit works fine on my nano and UNO.   On the Seeeduino, I cant see any serial output in the Arduino monitor and again, no MP3s start.

void setup()
{
  mySoftwareSerial.begin(9600);
  Serial.begin(115200);
  
  Serial.println();
  Serial.println(F("DFRobot DFPlayer Mini Demo"));
  Serial.println(F("Initializing DFPlayer ... (May take 3~5 seconds)"));
  
  if (!myDFPlayer.begin(mySoftwareSerial)) {  //Use softwareSerial to communicate with mp3.
    Serial.println(F("Unable to begin:"));
    Serial.println(F("1.Please recheck the connection!"));
    Serial.println(F("2.Please insert the SD card!"));
    while(true);
  }
  Serial.println(F("DFPlayer Mini online."));

Is this a known issue or any known fix?

Thanks

8BitMark, 2020-12-17:
I'm having the same issue but with an ATTINY 85.  Works fine on the Uno, just not on the ATtiny.

Another post here "failure to initialise" suggests change the timeout to 1000ms just prior to the begin command and another to add an extra parameter of FALSE to the begin command.

I'm going to get this fix on the attiny tomorrow...

samazaphikel, 2021-03-20:
Do you have a solution? I have the same problem on Attiny.

> I'm having the same issue but with an ATTINY 85. Works fine on the Uno, just not on the ATtiny.
>
> Another post here "failure to initialise" suggests change the timeout to 1000ms just prior to the [...]

marcdraco, 2021-10-11:
Same on ESP32 also. Very troubling. I'm investigating as there seem be to be a number of us. The code (prima facie) seems portable but I've barely given it more than a casual glance.

KonradStuewe, 2021-11-04:
I had the same problem with an ESP8266 and ESP32. Both only displayed "Unable to begin" while it worked flawlessly on the UNO.

The only way I could make it work on the ESP8266 (haven't tried the ESP32 yet) was by using hardware serial, so the TX and RX pins. This does work but comes with a few drawbacks.

For one, you have to disconnect the RX and TX pins every time you want to flash it, since USB also uses those and the DFPlayer gets in the way of that.
Secondly, you can't use serial monitor to debug anymore, since USB again uses the same ports. You could in theory use the TX1 serial pin from the ESP to send data to the PC, but you would net a UART to USB adapter for that, which I don't have.
Instead, I used the onboard LED with different blink patterns to debug my code.