DFRobot/DFRobotDFPlayerMini #33: Dfplayer won't start when volume is changed in Void

nielsdefize, 2021-04-11:
I am building a Wake Up Light alarm clock. During the alarm the DFplayer should start at a low volume and should slightly increase. 
Strangely whenever I try to change the vlolume in the function of the alarm, there is no sound coming out of the player.
Any idea how I can fix this?
The alarm function  code:

void playAlarm()
{
  myDFPlayer.volume(2); // Whenever this line is removed it does work
  myDFPlayer.play(1);
  brightness = 30;
  analogWrite(led, brightness);
  delay(5000);
  myDFPlayer.volume(6);
  brightness = 60;
  analogWrite(led, brightness);
  delay(5000);
  myDFPlayer.volume(8);
  brightness = 90;
  analogWrite(led, brightness);
  delay(5000);
  myDFPlayer.volume(10);
  brightness = 120;
  analogWrite(led, brightness);
  delay(5000);
  myDFPlayer.volume(15);
  brightness = 5;
  analogWrite(led, brightness);
}

patrickatkeylogic, 2021-04-22:
Is the blue LED active on the DFPlayer board? Could it be that '2' is just too low to hear?

nielsdefize, 2021-05-03:
Thanks for the reply. The DFPlayer is working.The LED (red in my case) is active and whenever I change the volume, nothing changes.

TiJof, 2021-05-18:
Same for me
If myDFPlayer.volume(X); is elsewhere than setup(), the playback doesn't start.

For me, sound is always at max level, now way to lower it.

J-L-Fan, 2023-10-24:
There are many problems with this library. For example, when you use readVolume after playing a piece of music, the value read out will definitely be incorrect.

The problem you encountered can be solved by adding delay after volume(2).
like :
myDFPlayer.volume(2);
delay(500);
myDFPlayer.play(1);

9H1LO, 2023-10-24:
avoid using delay(), its messy, use millis()

> There are many problems with this library. For example, when you use readVolume after playing a piece of music, the value read out will definitely be incorrect.
> 
> The problem you encountered can be solved by adding delay after volume(2). like : myDFPlayer.volume(2); delay(500); myDFPlayer.play(1);