DFRobot/DFRobotDFPlayerMini #34: Cannot connect DFPlayer to Arduino Nano 33 IoT

ilvyanyatka, 2021-05-05:
Hello,
I am trying to connect DFPlayer to Arduino Nano 33 IoT.
I believe I cannot use SoftwareSerial there, but can use HardwareSerial Serial1, connection TX-RX and RX-TX. I also use external power supply for 5V, but it seems to be irrelevant. Player cannot be found with or without it (when Arduino is just connected to PC by USB).

The wiring
[image]

The code
`#include "DFRobotDFPlayerMini.h"

// Create the DFPlayer Mini object.
DFRobotDFPlayerMini myDFPlayer;

void setup()
{
  Serial.begin(115200);

  // Initiate DFPlayer Mini on the second serial port on RX0 and TX1 pins.
  Serial1.begin(1200); // tried 9600 as well
  while(!myDFPlayer.begin(Serial1))  { Serial.println("Not Connected!"); }

  Serial.println("DFPlayer Connected!!!");
  myDFPlayer.setTimeOut(500); //Set serial communictaion time out 500ms
  //----Set volume----
  myDFPlayer.volume(30);  //Set volume value (0~30).
  // Set EQ
  myDFPlayer.EQ(DFPLAYER_EQ_NORMAL);
  // Set the SD Card as default source.
  myDFPlayer.outputDevice(DFPLAYER_DEVICE_SD);
}

void loop()
{
  myDFPlayer.play(1);
  delay(1000);
  myDFPlayer.play(2);
  delay(1000);

}

All I see in Serial Monitor is “Not Connected” messages.

Is there something wrong with my wiring? With the code? Any suggestions are appreciated.
TIA`

Jens869, 2022-01-17:
try 
myDFPlayer.setTimeOut(1000); //Set serial communictaion time out 1000 ms

same problem as with ATMega2560

Jens869, 2022-01-17:
The library was developed for use with mySoftwareSerial and unfortunately not optimized for a hardware serial. 
But the hardware serial also generates less CPU load.

Seikonolff, 2022-03-10:
Hi, Did you manage to make it work ? I have the same issue with arduino nano 33 iot. It is kind of anoying...

Jens869, 2022-03-10:
still on software serial. I have to change the code and open the box to change the pins. but I dont want to do that because it work on software serial...
But i have hardware serial 3 still available ...

Seikonolff, 2022-03-12:
Silly question but have you put any SIM card on it when running the code ?