DFRobot/DFRobotDFPlayerMini #50: stuck on play

plutphil, 2023-06-30:
I don't know why but with the new version 1.0.6 my code gets stuck on play. I use an ESP32. I tried switching between Serial2 and SoftwareSerial - no difference.

ontheplane, 2023-07-08:
I have exactly the same issue using Arduino NANO. Had me changing out the sound cards to no avail before I realised it was a library update. Moving back to 1.0.5 fixed it.

laurentmas, 2023-07-13:
Hello, I have the same issue, the New library goes to infinite loop if you set ack to true and you try to stop a music. i open an issue with my findings

donjoergo, 2023-10-10:
I also have this issue and changing back to 1.0.5 also fixed it for me.

ontheplane, 2024-07-26:
This still has not been resolved insofar as my library keeps trying to update to 1.0.6. Is there a way to a particular update or is it all or none?

drf5n, 2024-09-13:
It's got to be one of the v1.0.6 commits on https://github.com/DFRobot/DFRobotDFPlayerMini/commits/master/

Maybe:

https://github.com/DFRobot/DFRobotDFPlayerMini/commit/ed07ad9da4edcde359fdfb566e9c09ff433eb867

or 

https://github.com/DFRobot/DFRobotDFPlayerMini/commit/a81a07c863f403cae84e37f2bf389bdd0a8e850f