DFRobot/DFRobotDFPlayerMini #54: Latest Version is not working (Hanging Arduino) when using multiple SoftwareSerial.

geraldcells18, 2023-10-31:
**The Arduino will hang when calling startSoundSerial() after I call stopSoundSerial()**
_But if you use v1.0.0 it working properly_

SoftwareSerial WifiSerial(DD5, DD6);
SoftwareSerial SoundSerial(A0, A1);

void startSoundSerial() {
  if (WifiSerial.isListening()) {
    WifiSerial.flush();
    WifiSerial.stopListening();
  } else {
    SoundSerial.listen();
  }
}

void stopSoundSerial() {
  if (SoundSerial.isListening()) {
    SoundSerial.flush();
    SoundSerial.stopListening();
  } else {
    WifiSerial.listen();
  }
}

constantinsta, 2023-11-03:
I have the same problem. I am using previous version 1.0.5 to stop hanging Arduino.

rikiriocma, 2024-03-24:
me too, back to version 1.0.5 after going crazy

gusmakous, 2024-04-09:
Hello!
I have a similar problem. esp32, Serial2. the program hangs when calling the .play() function and only if the TX pin on the DF player is not connected to the RX pin on the ESP. As soon as I connect both data lines, everything works as it should. Rolling back to a previous version helped. Also, in setup, I use the command myDFPlayer.setTimeOut(500); but when the 2nd dataline is disconnected, the program hangs when accessing the player.

Hereux, 2024-11-02:
I just spent something about 8 FUCKING HOURS debugging someones code just to find out that the library does not work. 
It should be illegal to not maintain a used library.